cribskip/esp8266_spa #1: Not compiling?

tmjo, 2020-08-14:
Hi cribskip! I found your repo when googling for an ESP library for Balboa. I've successfully done a few ESP8266 projects before, but I'm no expert. When I try to compile and upload your code to my ESP, it fails to compile with the following messages:

`balboa_helper:1:14: error: 'CircularBuffer' was not declared in this scope
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
              ^
balboa_helper:1:36: error: expected primary-expression before ',' token
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                                    ^
balboa_helper:1:43: error: 'data' was not declared in this scope
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                                           ^
balboa_helper:1:47: error: expression list treated as compound expression in initializer [-fpermissive]
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                                               ^
esp8266_Spa:53:16: error: variable or field 'print_msg' declared void
 void print_msg(CircularBuffer<uint8_t, 35> &data) {
                ^
esp8266_Spa:53:16: error: 'CircularBuffer' was not declared in this scope
esp8266_Spa:53:38: error: expected primary-expression before ',' token
 void print_msg(CircularBuffer<uint8_t, 35> &data) {
                                      ^
esp8266_Spa:53:45: error: 'data' was not declared in this scope
 void print_msg(CircularBuffer<uint8_t, 35> &data) {
                                             ^
balboa_helper:1:21: error: 'crc8' declared as an 'inline' variable
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                     ^
balboa_helper:1:21: error: redefinition of 'uint8_t crc8'
balboa_helper:1:9: error: 'uint8_t crc8' previously defined here
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
         ^
balboa_helper:1:21: error: 'CircularBuffer' was not declared in this scope
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                     ^
balboa_helper:1:43: error: expected primary-expression before ',' token
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                                           ^
balboa_helper:1:50: error: 'data' was not declared in this scope
 inline uint8_t crc8(CircularBuffer<uint8_t, 35> &data) {
                                                  ^
`

It's probably me or my setup, but I would very much appreciate if you could guide me on the way. Using latest Arduino IDE (1.8.13) and already included the required libraries. If I can get it running I would be more than happy to contribute with testing and further development.

Thx.

cribskip, 2020-08-14:
Hi tmjo,

thanks for pointing out. I've seen the code is incomplete. I'll fix it
tomorrow and please check again.

Best regards

tmjo, 2020-08-15:
That's awesome! Thanks!!

cribskip, 2020-08-15:
Ok, so I've updated the code to keep up to my local copy. Fixed the missing file and referenced the dependencies (Please install from the Arduino IDE's library manager)

tmjo, 2020-08-15:
Thank you! I tried the new file, but still get the same compilation errors. Read something about the compiler being weird sometimes when it comes to inline functions. Do you know if you have setup anything special to make sure it recognizes the balboa_helper correctly?

I'll do some digging. I should be able to compile it if you are running it, so it must be something with my setup vs yours I guess.

tmjo, 2020-08-16:
Hi again! I think I kindof found the issue. Could it be that your local project folder is "esp8266_**S**pa" with capital S while the github repo is "esp8266_**s**pa"?? Perhaps github is forcing lower case names? Don't even know how I got onto that track, but my folder was all lower case while the main file was with capital letter. This seems to give me the errors that I described in my initial post. Both renaming the folder to capital S or renaming the main file to lower case s removed all those funny compilation errors. Seems strange to me, Arduino IDE was already giving me an error since my initial folder name was directly from the Github download and included "-master" in its name, so I changed to just "esp8266_spa", but it didn't at all complain about the upper/lower case thing. I was able to reproduce the issue both ways several times.

Once that was solved, I still got one compilation error....
```
esp8266_spa:160:24: error: 'class String' has no member named 'toDouble'
     double d = payload.toDouble();

```
.... which seems strange. I googled and found the _toDouble()_ everywhere, but also stumbled over some suggestions of using _toFloat()_ instead.  Replaced line 160 from  _double d = payload.toDouble();_ to float _d = payload.toFloat();_ and it compiles fine. This seems odd to me, but could it be that it has been removed somehow from the Arduino?? As mentioned before, I'm using the latest 1.8.13 of Arduino IDE - do you mind me asking which version you are using? Couldn't find any references on google on if/why it is removed, and it seems odd if it is, so I don't know....

Anyway, now it is compiling, so I'll try it live on my tub tomorrow! Thx.

cribskip, 2020-08-16:
Thank you for your investigations, tmjo :-)
Always great to see helpful people like you.

I'm using Arduino 1.8.12 with esp SDK 2.7.2.

Your guess concerning different names is right - I'm using OpenSpa and renamed it for clarity on github. There I may have introduced the naming trouble. I'll fix it ASAP.

Good luck and keep safe, turn off power.

The code has a problem but mitigatable: https://github.com/cribskip/esp8266_spa/issues/2

tmjo, 2020-08-16:
Great! I thought my esp sdk was updated, but turned out I was on v.2.3.0. Have a vague memory of having to downgrade some time back due to some problems on a previous project, but now I updated to latest v2.7.4 and the toDouble() is working fine. Must have happened something there in between versions.

Another thing, completely unrelated to the "issue", but I'll give it a shot anyway: I just found out that my Balboa panel is connected with a 8-pin Molex and not 4-pin as documented on the ccutrer/balboa_worldwide_app repo. Tried to measure, but found 10V instead of 12V and several 5V instead of 2-3V. Need to investigate a bit more before trying to connect. Any chance you've seen anything about the subject? You have the 4-pin Molex?

tmjo, 2020-08-16:
Closing the issue, thanks for your help!

closed 2020-08-16 by tmjo