cribskip/esp8266_spa #3: Problems with compiling sketch

Steem147, 2020-09-01:
Best cribskip,

I have the following problem when trying to compile the sketch. I have add the circularbuffer 
and also the pubsubclient, i have selected the ESP8266 Wemos D1 R1.

The following is the compiler log:

>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>
esp8266_spa.ino: In function 'void loop()':
esp8266_spa:252:16: error: 'ID_ack' was not declared in this scope
         ID_ack();
                ^
esp8266_spa:258:20: error: 'ID_request' was not declared in this scope
         ID_request();
                    ^
esp8266_spa:278:20: error: 'rs485_send' was not declared in this scope
         rs485_send();
                    ^
exit status 1
'ID_ack' was not declared in this scope
>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>/>

What could be the problem?

Best regards,

Steem

cribskip, 2020-09-02:
Hey Steem,

those functions are in balboa_helpers.ino.

Did you put the file beside the main ino file? 
What version of arduino ide and esp8266 sdk are you using? 

Best, Sascha

Michael0yodi, 2020-10-15:
Create a new folder and copy both *.ino files there and open the esp8266_spa.ino in Arduino IDE via File -> open and add the "balboa_helper.ino" file via Sketch -> Add File it will read it and compile just fine

cribskip, 2021-02-17:
Is this issue still valid?

srasmond, 2021-03-31:
I followed the instruction of Michael0yodi as well (for a Lolin (WeMos) D1 Mini Pro) and it worked for me. So I think the issue is resolved when following this procedure.
In addition, for Mac users, the following might be intresting when encountering upload issues with Arduino IDE: https://forum.arduino.cc/index.php?topic=702144.0 (comment of "juanssl" provides the fix)