cribskip/esp8266_spa #4: Topic only shows me Spa/node ON

Magfel, 2020-10-15:
Hi,
I like this project, thanks a lot.
I did all you recommand but when when I include MQTT client, the only command founded is "Spa/Node" with the status ON.
Can you explain me why ? My jeedom seems communicate right with the Wemos D1 Mini pro but not with th tub.

cribskip, 2020-10-16:
Hi,
thank you for the heads up. 
Please subscribe to Spa/# as you will then see debug messages on /debug. 

In the code are several places with comments to higher the verbosity. Please test these.

Which spa controller are you using? There are differences between controller versions which may cause a problem. 

Wemos d1 mini is right.

Magfel, 2020-10-16:
I have a Balboa BP2100G1.
When I suscribe to Spa/# I have only the Spa/node information with a ON status.
Once, I've seen the debug information witch returned -2 ou -4 value, I think.

cribskip, 2020-10-17:
I have the BP2100G0. 

| I've set the DC-DC converter to 5V to power the Wemos D1, is it right ?
Perfect :-)

| Is it possible A and B wires inverted ?
That may be the case, please swap them and try again.

Debug is used for several informations. When reconnecting, it publishes the MQTT state before the reconnect (e.g. TImeout etc.) [Docs](https://pubsubclient.knolleary.net/api#state)

Magfel, 2020-10-26:
OK, I have a BP2100G0 too, it was a mistake from me cause I'd downloaded the BP2100G1 datasheet.
Now, it works fine since I've change the Wemos card to an original WEMOS D1 mini pro, I don't known what was the problem with the first one.
I've added a few lines of code to have the possibility of uploading the sketch through the wireless connection. 
And I have put all the electronic components in a 3D printed case 
[image]

cribskip, 2020-10-26:
Nice to hear it's working for you also :-)

HTTP-Updater ist already implemented: https://github.com/cribskip/esp8266_spa/blob/b1cb9cef590acb9f651b07251ffd164370011b81/esp8266_spa.ino#L209

Sweat case, could you please post a picture from your setup inside?

Magfel, 2020-10-26:
[image]
Is is a beta version, I'm going to do one larger.
How do http-updater works ? When I go to the page \\192.168.0.164, I have a blank page with "Not found: /" written on the top left.

cribskip, 2020-10-26:
Use http://ip/update
If asked, use user admin, blank password

Neat setup compared to mine 😅

Magfel, 2020-10-27:
Thanks :-)
_Use http://ip/update_ it works, but I prefer using OTA version.
I know why my first "Wemos" card didn't work. On these cards, the pin D1 is the GPIO5 !!!

EmmanuelLM, 2020-12-24:
> Thanks :-)
> _Use http://ip/update_ it works, but I prefer using OTA version.
> I know why my first "Wemos" card didn't work. On these cards, the pin D1 is the GPIO5 !!!

Hi @Magfel I am experiencing the same issue as you but don't understand your last comment as D1 is GPIO5 on my Wemos D1 Pro?? :-/

Magfel, 2020-12-24:
EmmanuelLM,
Try a little program using D1 linked to a LED and test it to be sure it's the real pin used by the program. It was the problem with a copy of the Wemos PRO.

EmmanuelLM, 2020-12-26:
I just checked this and D1 is indeed D1 ... I double-checked my chip as well, which seems to work.
It definitely feels like a hardware issue rather than a software one, but I cannot understand why the program would only post ON and not post the other messages right after it....

cribskip, 2020-12-26:
Excuse me, I'm late back to the discussion ;-)

@EmmanuelLM Did you see this Hint in the code and test it out? Please also post a picture of your setup, thanks
https://github.com/cribskip/esp8266_spa/blob/5f867a4486b858caddb717e2fb8187c5919024d6/esp8266_spa.ino#L245

EmmanuelLM, 2020-12-26:
I did yes, when I get something, it is 254... in fairness, I wonder if my cable is at fault, so I have ordered a new one and will report back when it arrives next week.

EmmanuelLM, 2020-12-30:
I am back with a new cable and still the same issue.... I also noted that the node timeouts every few seconds from the mqtt broker (Mosquitto) and that going to http://ip/update also produces a ERR_CONNECTION_TIMED_OUT - it makes me wonder if the problem is perhaps linked to the network? looks like the node may be trying to reconnect on a regular basis hence the constant Node ON messages... but I don't fully get why I don't receive the debug messages at that point as well...

EmmanuelLM, 2020-12-30:
One other question / possibility why it could not work - I see the Max485 breakout board is a 5V logic vs the wemos d1 mini pro is 3.3v logic, which makes me wonder if the RX/TX are 'overdriven' at 5v, which could cause issues...?

pengemetal, 2020-12-30:
> I have a Balboa BP2100G1.
> When I suscribe to Spa/# I have only the Spa/node information with a ON status.
> This night, the debug commande was added. It returns a -4 value.
> I've set the DC-DC converter to 5V to power the Wemos D1, is it right ?
> Is it possible A and B wires inverted ?

Hi Everybody,

I am very happy to find this topic! :). I have the same Balboa BP2100G1 control panel. Could you explain where i can plug Wemos D1 to the panel? The J34 wifi socket? And the DC-DC converter, is it coming from Led 12V? Could you give me some more information about wiring? Thx in advance

cribskip, 2020-12-30:
@EmmanuelLM 
Ok, so your cable seems to be good. Maybe the wifi distance is your problem but in this case, you may still see the RECONNECT message. What's the Reconnect reason when the node reconnects? How many millis has the node been running? Or do you ONLY see Node Online?

@pengemetal 
Yes, you are right, the hook up documentations is rather sparse ;-) The stock display has a 4 pin molex-type connector with +12...15V, GND and A+B. All connections are on the wire and needed. Please also see the referenced https://github.com/ccutrer/balboa_worldwide_app/wiki#physical-layer for further details. The LED 12V supply for the RGB lighting in the spa is NOT being used.

Testing for correct cabling to the SPA, you can also hook up a USB TTL UART to the RO Pin of the Max485 and bridge RE/DE to GND. You should see plausible data at 115200 8N1.

EmmanuelLM, 2020-12-30:
Hi, I don't see the reconnect message, and spa/debug reports -1 (when it does), with a spa/rcv of 254.

cribskip, 2020-12-30:
Just for testing, please swap a and b

pengemetal, 2020-12-30:
> @EmmanuelLM
> Ok, so your cable seems to be good. Maybe the wifi distance is your problem but in this case, you may still see the RECONNECT message. What's the Reconnect reason when the node [...]

Thank you for your answer! Just one more question: Could you help mi about relays wiring? What are they doing? Thx

EmmanuelLM, 2020-12-30:
> Just for testing, please swap a and b

Will try, i'll also hook up a usb ttl for kicks!
Did you see my other comment about the max485 being a 5v chip?

cribskip, 2020-12-31:
@EmmanuelLM 5v is OK for newer chip revisions since a few years. Unless you took a trip to the museum you should be fine.

@pengemetal the relays in the picture were to switch on external lamps. Then for J29 (power shedding of the heater) meanwhile, they are near to useless because I've attached a good and cheap heat pump and disconnected the internal heater (code to come)

Magfel, 2020-12-31:
> I am back with a new cable and still the same issue.... I also noted that the node timeouts every few seconds from the mqtt broker (Mosquitto) and that going to http://ip/update also produces a [...]

Hi EmmanuelLM, Does the wifi signal strong enough ?

EmmanuelLM, 2020-12-31:
Hi, Wifi is good, i am a few meters away from the router, I have also tried switching A & B with the same result.
The one other difference I can think of vs the native code is the change mqtt.connect to take into account authentication so mqtt.connect("EspSpa", "login", "password")

I have a usb - ttl coming later today so will do a step by step to ensure my tub is actually producing something (the display works so I think the answer is yes) and work my way backwards to diagnose the issue... if helpful to anyone, this device seems to be able to do it all! https://www.ebay.co.uk/i/202172720183

EmmanuelLM, 2020-12-31:
News from the front: it works!!!!
The issue was the RS485 to TTL module that I was using, I ended up swapping it over to a "ARCELI TTL To RS485 Adapter 485 Serial Port UART Level Converter Module 3.3V 5V" and it now works flawlessly. I will push over some changes to the doc to add a diagram and help other people who might come across here.

The only last issue I have is the wemos d1 mini pro does not 'boot' by itself with the current power rail, I need to first connect it to a usb power supply and then it works as expected (including resets...) - how strange...

cribskip, 2020-12-31:
@EmmanuelLM thanks for the update and great it works. Rather strange with the power supply. PR welcome 😅

EmmanuelLM, 2021-01-03:
Hi, I have started making changes to the code, mainly to deal with the following:
- remove the need for D1 with the RS485/TTL converter
- add support for Home Assistant autodiscover support
- deal with spurious temperature readings (not sure if that was just me?)
- shuffle the mqtt message structure and add more debug info
I migrated the files to PlatformIO as I find it easier to develop on that - this does mean the source file is not in the same location as yours and I have not yet figured out how to make a pull request from alternative folder locations... once I do, I will send PRs
You can see changes here in the meantime: https://github.com/EmmanuelLM/esp8266_spa/blob/master/src/esp8266_spa.ino
If you go to the root of the fork, you will see I also updated the documentation, I will send a PR for that