cribskip/esp8266_spa #5: "Nothing to send" not handled correctly?

irekzielinski, 2020-11-14:
Hi. Thanks for your project, it's a great starting point!
While comparing code to the specification I noticed that it's not handling "Nothing to send".
According to: https://github.com/ccutrer/balboa_worldwide_app/wiki#nothing-to-send

"Nothing to send" should be sent when "Clear to Send" is received but we have no update.
Right now code will send "Toggle Item Request" instead with byte 0x00 - this does not seem to be correct.

Here is the amended code (send == 0x00):
```
       if (send == 0xff) {
          // 0xff marks dirty temperature for now
          Q_out.push(id);
          Q_out.push(0xBF);
          Q_out.push(0x20);
          Q_out.push(settemp);
        } else if (send == 0x00){
          // A Nothing to Send message is sent by a client immediately after a Clear to Send message if the client has no messages to send.
          Q_out.push(id);
          Q_out.push(0xBF);
          Q_out.push(0x7);
        } else {
          // Send toggle commands
          Q_out.push(id);
          Q_out.push(0xBF);
          Q_out.push(0x11);
          Q_out.push(send);
          Q_out.push(0x00);
        }
```

Let me know your thoughts.
Cheers,
Irek

cribskip, 2020-11-15:
Thanks Irek, I'll try it out when time (and weather ;-)) permits.

I've deliberately done it this way as it's the same behaviour of the original controller in my case.

Are you also using a balboa spa? Which controller then?

Best,
Sascha

irekzielinski, 2020-11-15:
Hi, interesting that your controller sends 0x11 in this case.
I have Balboa VL600S and it sends 0x7. Possibly both are OK.
Thanks!
Irek

ps. noticed that RO pin is directly to RX in Wemos. I think RO can go up to 5V which is higher than spec for Wemos. I put a level shifter in my implementation (voltage divider would also do). There are people that have wemos working on 5V input pins for years, other say board can get damaged over time. Did you see any long term issues with serial or board stability?

cribskip, 2020-12-26:
Hi,

thanks for your feedback on the VL600s behaviour.

No, it's up and running smooth for about three months. Its up to the chip revision if they can stand 5V in or not. Older revisions did not while newer ones do.

Did you build the project yourself?

Cheers,
Sascha

Magfel, 2020-12-31:
Hi,
I've a similar problem. I can send orders to the tub about 2 hours after powered on the Balboa controller. After this laps time, I can only receive messages and it works all the time. If I have to send another message passed this 2 hours, I have to swith off and on the controller Balboa.  It is probably caused by the byte 0x11 instead of 0x07 as says irekzielinski.
What do you think  ?

irekzielinski, 2020-12-31:
Hi Sascha. Yes I build the project myself - your code was a great help! Thank you.
 
> EDIT: Just pasted the code you thanksfully provided in and noticed a missing 0 with the message type code. It's 0x07 instead of 0x7. Thank you :)

There is no difference between 0x07 and 0x7 - it is the same value of 7. Might be worth doing this for the esthetic reasons but compiler will generate the same code in both cases.

Kind reagrds,
Irek

irekzielinski, 2021-02-17:
> Hi,
> I've a similar problem. I can send orders to the tub about 2 hours after powered on the Balboa controller. After this laps time, I can only receive messages and it works all the time. If I have [...]

I heavily modified the code - but main difference is that I send 0x11 and use level shifter to convert from 3.3V to 5V when interfacing with RS485. My controller is rock solid - working for months without any reboot, lost connection or change of ID (ID is assigned on first data exchange with Balboa mainboard).
I recommend to do both modification.

cribskip, 2021-02-17:
Thanks for your feedback and optimization recommendations @irekzielinski ! Very apprechiated!

Did you try the current version yet? Very cool with the homeassistant auto detection and configuration parser! Great work from @EmmanuelLM

closed 2021-11-26 by irekzielinski