cribskip/esp8266_spa #7: No Topic messages

w4Rd3n, 2021-02-03:
Hello.
Thank you all for the great work on this.
I have some problems with the fork of Emmanuel:
https://github.com/EmmanuelLM/esp8266_spa

I use a Wemos D1 mini  Pro
and this config:
```
#define VERSION "0.32"
#define WIFI_SSID "TP-Link_F526"
#define WIFI_PASSWORD "******"
#define BROKER "192.168.1.170"
#define BROKER_LOGIN "loxberry"
#define BROKER_PASS "********"
#define AUTO_TX true //flag for the RS485/TTL chip - i.e. do we need to pull D1 or not....
```

At the moment i don't have the spa connectet to the RS485  TTL transceiver.
But anyway i should get some mqtt messages, no ?
I am powering the wemos on the 5V input with my DC-DC converter

So far Wemos is connected to my wifi but i dont receive any spa  message on my broker.
i am subscribed to Spa/# topic.

Here is a photo of the HW:
[image]

w4Rd3n, 2021-02-03:
I don't use external Antenna on the D1 mini for now. Ping is working but usually 1 of 4 packets are lost wheni execute the command.
The Webserver is not working. it keeps saying connection timeout and then reloads page

cribskip, 2021-02-03:
@EmmanuelLM s fork is tested with me in PR #6 but up to now, my tests were not successful besides the code looking great.

Maybe you should try my current code first to debug potential connectivity issues as it publishes when the wemos connects to the broker.

EmmanuelLM, 2021-02-03:
My fork is "work in progress" ;-)

From my experience, I had issues with the RS485 to TTL transceiver and found [and I do not know why] that the device would just not work if it didn't receive messages (c.f. issue #4)...

I agree with @cribskip - test his code first :)

p.s. #define AUTO_TX true -> this should be false with your transceiver

w4Rd3n, 2021-02-04:
Thanks for your inputs.
For now i only powered the board without the spa connection.
i was assuming i ll get some mqtt messages also without the spa connected.
I will let you know how this continues :)

Michael0yodi, 2021-06-01:
Did you ever find any solution I've similar issue, I can see that it reports to my broker

1622540317: New connection from 172.16.0.121 on port 1883.
{"result": "ok", "data": {}}1622540317: New client connected from 172.16.0.121 as Spa8861 (p2, c1, k10, u'mqttuser').
1622540332: Client Spa8861 has exceeded timeout, disconnecting.
1622540338: New connection from 172.16.0.121 on port 1883.
1622540338: New client connected from 172.16.0.121 as Spa8852 (p2, c1, k10, u'mqttuser').

But no matter if connected or not to the spa I never get any data via mqtt
webserver on D1 just give me this "Not found: /" not sure if expected?
 
I've it running on usb power without using the dc-dc converter, but since ground is not connected on the rs485 I guess it "should" work, rxd is flashing so looks like data is received

[image]

EmmanuelLM, 2021-06-03:
Hello, a few things as it looks like you are nearly there.

If you are powering off usb (or anything else), you need all GND to be connected together otherwise the signal 'floats' - likewise, with the hot tub GND.

if you go to the webserver, go to http://you_ip/update - nothing shows in the root as you have seen.

pato-loco, 2021-06-14:
@EmmanuelLM what do you mean with "all GND to be connected together"?

EmmanuelLM, 2021-06-14:
I meant that the following GND connections should all be connected together:
- power supply GND
- circuit GND
- Hot Tub connector GND

Michael0yodi, 2021-07-04:
I've tried to make sure that gnd are common with the spa using a dc-dc converter, but it still behaves the same so not sure where to start troubleshooting, any ideas are welcome

pato-loco, 2021-07-04:
> I've tried to make sure that gnd are common with the spa using a dc-dc converter, but it still behaves the same so not sure where to start troubleshooting, any ideas are welcome

put (-) from the DC-converter "in" and (-) from the DC-converter "out" directly together.

Michael0yodi, 2021-07-04:
should I also connect the GND on the RS485 transceiver? I've used RS485 before with no common ground so its a bit strange that its not working at all, could of course be something else also..

tonyfitzs, 2021-09-28:
hi there, I am also having issues with MQTT. as reported above, I can see in the logs of my broker that the D1 Mini running this code is trying to connect to the brocker and its trying to send a message as follows

1632863671: New connection from 10.110.x.x on port 1883.
1632863671: New client connected from 10.110.x.x as Spa1 (p2, c1, k10, u'test').
1632863665: Client Spa1 has exceeded timeout, disconnecting.

but the subscription isnt ever completed before it disconects. i am wondering if the code I am using is the right version (VERSION "0.35"), because the webpage, dosnt do anything either, and based on above it looks like it is suposed to. if I try to hit http://10.110.x.x/update I get a page time out. perhaps I am misunderstanding the purpose of this url, because the documentation has a few gaps, is this an OTA address rather than a webpage that shows you information.

EmmanuelLM, 2021-09-29:
Hey there, thanks for trying the project.
In my experience with this type of issue, the problem ended up being the
RS485 TTL transceiver - the chip reboots itself when it doesn't receive any
messages on the bus, so the behaviour below is expected.

If you have not already, I highly recommend getting the ARCELI TTL To RS485
Adapter 485 Serial Port UART Level Converter Module 3.3V 5V for the simple
reason you don't need to mess about with levels.

If you already have this chip, you can buy a cheap rs485 TTL usb debugger
that you can hook up to your laptop to check messages are coming through.

Good luck!

P.s. the OTA page is just for updates but it won't be available as the
device reboots itself.
P.p.s. there is a newer firmware on my fork

tonyfitzs, 2021-09-29:
Hi Emmanuel,

Thank you for taking the time to get back to me. I have high hopes for this project. So first, let me clarify a few things; I have purchased the hardware that you recommended on your GitHub page. That means I have a D1 mini pro, and I also have an  ARCELI TTL To RS485 (https://www.amazon.co.uk/ARCELI-Adapter-Serial-Converter-Module/dp/B07DN115BZas) before I go out and mess around under my house to connect all this up to my spa (My spa is built into the floor of my rear entertainment area) I want to make sure this all works on my workbench. So I also purchased a cheap RS485 to USB (https://i.ebayimg.com/images/g/~n0AAOxyItRTefGv/s-l500.jpg) as well.

I am also going to create a small app to help me with testing using C#. This will basically replicate the spa's settings and generate a simulated message as if your spa was connected to the communication port. Once I have that all working, I will share the project because I think it will be useful for others following after us. So as the first step in that process, I am trying to get basic communications up and running between the Rs485 dongle and the D1 Mini project. This is where I am having a lot of trouble, and im trying to narrow down the variables to make sure it's not my hardware.

So what's happening is I have the d1 mini, and the ARCELI TTL To RS485 all set up, I have installed your code (VERSION "0.35") onto my d1 mini, and I have connected the A and B data lines between the TTL To RS485 and the Rs485 dongle. Once the D1 boots up after loading the code, I open the terminal to see what it is sending (this is the screenshot on the left below) I also open a terminal against the USB port (on the right), and I can clearly see that the D1 is sending messages to the USB dongle, I send a sample message (7e 20 ff af 13 00 00 ff d 2a 00 00 67 67 00 04 00 00 00 00 00 00 00 00 00 66 00 00 00 78 00 00 52 7e) I can see that the D1 mini is responding with

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000731e0
~ld
fpm close 3
mode : sta(a0:20:a6:0d:d6:ba)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt

So far, so good.

***@***.*** ***@***.***

The issue I am now having is that I can't seem to get it to connect to my MQTT broker,

In the broker logs, I get this message repeated over and over again
1632903370: New client connected from 10.110.10.147 as Spa1 (p2, c1, k10, u'test').
1632903386: Client Spa1 has exceeded timeout, disconnecting.
1632903391: New connection from 10.110.10.147 on port 1883.

And I can't see a topic for the spa being generated. Based on what I can see in your code, I think it should be reporting something like Spa/<topic>/<state> but its not.

Anyway, if you can help, I would appreciate it.

Also, I would like to help you out with improving the readme you have created because there are a lot of gaps in the explanation, which makes it a lot hard when you are having trouble like I am. Once I have this all up and running, I would be very happy to re right this for you and send it to you so you can post it if that's of value to you.

Thanks for your time

EmmanuelLM, 2021-09-29:
One quick question: the system is not yet hooked up to the Spa, correct? If
that is the case, I don't think the system will work because it will reboot
itself when it doesn't receive messages after a while.

tonyfitzs, 2021-09-29:
Yes that's correct.

Ok that makes sense. In that case I will get my little balboa Spa simulator working and revert...

But what about the mgtt question.

Michael0yodi, 2021-09-29:
If you dont want it to reset you can just comment out the hardreset(); or set #define PRODUCTION false
I did however never manage to get any mqtt messages to arrive beside the debug/connect ones, I thought my rs485 adapter might be broken, (same version as yours) but never got around to order another one, so please let us know if you make any progress

EmmanuelLM, 2021-09-29:
Same answer I am afraid, topics only get published once the system has
retrieved the configuration from the spa.

tonyfitzs, 2021-09-29:
Very well…

Thank you

StevieWW, 2024-04-09:
Make sure you use a tranceiver with a MAX13487 Chip and nit just MAX485. This was my problem since it seems that the layout of the linked board has been changed in between.