cribskip/esp8266_spa #9: RS485 bus transceiver wiring

pengemetal, 2021-02-05:
I have this type of RS485 bus transceiver, could you help me about wiring? Thx
[image]

EmmanuelLM, 2021-02-14:
Evening, I had a lot of trouble with this transceiver so ended up getting a ARCELI TTL To RS485 Adapter 485 Serial Port UART Level Converter Module 3.3V 5V, so I would recommend getting one of those.
The main differences are:
- it natively runs at 3.3v or 5V
- you don't have to mess about with pulling D1 to transmit data
- it has some flashing lights that shows data being received / sent [helpful for debugging connections]

pengemetal, 2021-02-15:
Thx, I'll try to order this  transceiver!

pengemetal, 2021-03-31:
Hi, All parts have arrived I would start to make the connection. One question: I haven't got molex cable or Y splitter, is it possible to cut the display cable and with some Wago make DIY y-splitter cable? How did you wire to the motherboard? Thx!

EmmanuelLM, 2021-03-31:
I mean... everything is possible and what you suggest is (in essence) a Y-slitter.
I got a mini Molex cable and connected to the spare port.

One question for you (and whoever else is ready this) - would there be any interest in buying a made-up system? Thinking out loud, I am thinking the whole system + 3d printed case would probably come to ~50USD or so?

EspenT, 2021-03-31:
Yes, for 50usd I would buy 😊 I´ve been following this thread forever and probably have everything I would need to make it, except time.

pengemetal, 2021-03-31:
Thx for the quick answer! I think it is a fair price for it!

EmmanuelLM, 2021-04-01:
Sounds like I need to have a look at getting the wifi hotspot working then!

cribskip, 2021-04-02:
@EmmanuelLM please add a connector for J2 (IIRC) to control the heater. That way controlling the circulation is possible and other cool things.

Currently I use a relay but an transistor should suffice

mopac, 2021-07-06:
I realise it is a little late for the original poster, but just for completeness, I show here my schematic and a few photos of how I used this RS485 board.
I use a D1mini power sheild to regulate the hot tub 12V supply down to 5V for the RS485 board and the D1 Mini Pro. (D1 Mini Pro has its own regulator from 5V down to 3.3V.
The RS485 board is 5V, while D1 mini Pro is 3.3V. All outputs from D1 Mini Pro work just fine with a direct connection. The RX input - RO output from the RS485 board is the only issue. The fix is a schottky diode in this signal line. Schottky diode because of the low forward voltage. I used a BAT43 diode.
Tx & Rx used for the RS485 board are also connected to the USB-UART converter on the D1 Mini Pro but there are 470 ohm series resistors so they can be overdriven.
I use GPIO13 & GPIO15 for enabling the Receiver and Transmitter respectively. This is because GPIO13 has a pull-up on the D1 mini and GPIO15 has a pull-down. So if software doesn't configure these GPIO, the RS485 board is disabled. The benefit is that Serial download still works.
The only annoying tweak is that R3 on the RS485 board needs to be removed. This is a pull-up resistor that conflicts with the pull-down resistor on the GPIO15 on D1 mini Pro.
The only changes in the code are to configure the two GPIO as outputs and set LOW in the 'Setup' function. And ensure both GPIO are set HIGH then LOW in the 'rs485_send' function
[image]
[image]
[image]
[image]
[image]
[image]
[image]

closed 2022-01-10 by cribskip