cribskip/esp8266_spa #11: Wemos keeps resetting

micv83, 2021-04-05:
I’ve tried with 2 different wemos d1 mini, but after it connects to WiFi it disconnects again crashes with wdt reset.

what Couple be wrong ?

cribskip, 2021-04-05:
Power Supply ok?
Please post a photo of your setup

micv83, 2021-04-05:
The resets occurs even when hooked up to the laptop via usb, with only the Wemos - no regulator or rs485 attached. I have also tried standalone setup with 12v psu through regulator with rs485 attached.
I do get connection attempts in mosquito in home assistant, so it does something before it crashes...

Like this:
1617639019: New connection from 192.168.1.78 on port 1883.
1617639019: New client connected from 192.168.1.78 as Spa1 (p2, c1, k10, u'mqtt').
1617639034: Client Spa1 has exceeded timeout, disconnecting.

EmmanuelLM, 2021-04-06:
Hi, in my experience the device resets when it doesn't receive messages from the hot tub (see: https://github.com/cribskip/esp8266_spa/issues/7#issuecomment-772675742).

I am afraid you're going to have to hook it up to resolve that one :)

micv83, 2021-04-06:
Thank you srasmond - this was the issue, it stays connected now, and I am more confident trying to connect it to the spa now 😊👍🏼

cribskip, 2021-04-06:
@micv83 Good luck then! What controller do you have?

micv83, 2021-04-07:
It’s a Balboa bp6013g1 hope it works 😊

srasmond, 2021-04-07:
> It’s a Balboa bp6013g1 hope it works 😊

I got the same. Works brilliant, I control it via OpenHAB (Mosquito as broker). Already quite some things you can monitor or change

EmmanuelLM, 2021-04-07:
Hey @srasmond - what other features would you like to see implemented? I am currently working on an access point mode, so I can start releasing binaries, which would make the initial setup much easier, but wondering whether there are other things I could also knock off people's wish list?

cribskip, 2021-04-07:
@EmmanuelLM setting filter cycles and wifi manager would be great.

I'm currently polishing my heat pump controller and write up of the setup and photovoltaik optimization 😉

micv83, 2021-04-07:
[image]

Managed to get it working, but it’s not all that stable at the moment, but I can see the WiFi signal is very poor, so I’ve ordered a wemos with external antenna connector hope that makes the switches work more reliable 😊

srasmond, 2021-04-09:
@EmmanuelLM I am still playing around with some settings. Currently, my main goal was to set off the heating of the hottub during the night (with a minimum temperature of 21° C, but that is only a concerning during winter). At the moment I am realizing this by setting the "target_temp" lower each night (via openHAB rule). I tried to just turn of the heater and thought it does seem to work, there seems to be 2 issues in this scenario: 
- If you don't use the hottub for some hours, the temperature will actually be a "null" value, which requires to turn on the jets for a couple of seconds.
- If you turn on the jets, it seems like no matter what, the heating mode is overruled and it heats up anyway.
- Because of this you cannot continuously compare the current temperature, nor have a trustworthy method of turning of the heater.

--> Setting the target temperature provides a more ensuring way that the hottub will indeed not heat during that amount of time and as a plus, it seems like in this case you get continously updates current temperature values.

Other things I am thinking about:
- Filter cycles, but I see that there is a pull request for this, this seems like a really nice addition! Thanks!
- Setting the time (needs to be done manually now after hot tub is turned off or changing of winter- to summertime) --> important for the filter cycles.
- Really nice would be if we had some way of controlling the amount of power that the heater is taking for example. Most of the time, I am using the hottub after 6 PM, so during the day, when I give the chance to heat up, it would be nice if I could spread out the amount of power, so we don't get peaks but a more slower heating up during sun hours (to take full advantage of my solar panels).
- Control over the lights: at the moment I can only turn the lights on or off, but it seems it always is using the light looping mode (so looping over each color) and on a fast pace... Maybe I am using it wrongly at the moment, but if not it would be nice if we could set the color (or looping on/off) and the tempo (in fixed steps probably) of the lighting. In that scenario, I can also imagine that we can read out the current light color value and for example pass it to other smart lights in the house or garden to get a really nice effect.
- The ability to control / turn off the info messages on the panel on the jacuzzi.
- Setting a "vacation" mode: I guess this can be done by setting "heatingmode" to off (=ready vs rest mode)? Or what is the difference between head_mode and heatingmode?

srasmond, 2021-04-09:
@micv83 Getting an external antenna is a good idea. The default is indeed not strong enough (in my case for example, the signal needs to get from inside its case, which is inside the hottub (close to the balboa controller), which stands outside in the garden, to inside my house - to the closest receiver. Considering also the extra interference by all the wiring inside the balboa controller and perhaps even a bluetooth device next to it, getting a decent antenna is the minimum. After that, you should be all good.

cribskip, 2021-04-11:
@srasmond I'll try to answer to your rather long text structured ;-)

Heater off during winter:
You're right, when the temp is low AND "Balboa M7" "Artificial Intelligence Blockchain Crypto" is turned OFF, you're getting 0 as current temperature as the temperature is unknown to the controller. When M7 is on, the tub starts the circulation just before invalidating the temperature for a few minutes and knows the temperature again.

When you start the jets, the tub goes into ready-in-rest mode: Meaning it's heating to the Ready-Temperature.

My current setup (which needs to be described more in detail ;-) which circumvents all those problems:
- J2 on the tub mainboard it connected to a relay (J2 controlles wheater the tub is allowed to turn on the heater)
- J2 closed: Heater allowed, J2 open: not allowed
- (I disconnected the immersion heater and installed an external controlled heatpump...)
- When the tub invalidates the temp, I simply open J2 and set the tub to Ready mode. Circ starts and a few minutes later I set mode to Rest.

Some benefits:
- Heating is controlled by the heat pump which is cheaper and more environmently friendly
- A new "circulation only" mode is possible which is great when tubbin' because we have a fountain which is linked to the ever switching circ pump in ready mode...

TL;DR: I need to get my documentation ready. But first: more photovoltaic :-D

srasmond, 2021-04-12:
@cribskip The J2 connector would be my alternative to put a wifi switch relais on if I would not get the controller to work.
Now that I can control the hot tub in many other ways, I don't see an immediate benefit from using the J2 connector, but I am curious on what you mean exactly with the "circulation only" mode?

At the moment I am using the "target_temp" since a couple of days, and I have to say it works really well this way. At 10 pm a script is setting the "target_temp" to 30°C and at 10 am a script is setting the "target_temp" to 38°C again. This also makes sure that the current temperature is always known (during heating up or cooling down moments), which is handy, because now I see how much the T is dropping during the night (at the moment it never drops below 34,5 °C).

When light control, time and power control (will most likely need to be done at input level, not via the controller) is added to my system, for me it will have everything I would ever need.

If I have some time this week I will try to put my system, configuration, use cases, etc. on paper. You can then use whatever you want from it in future documentation.

pengemetal, 2021-04-15:
Hi! I use wemos D1 mini (not pro) how can I check the wifi connection? I can't see nothing in MQTT explorer, I think Wemos do not connect to wifi (SSID and password is correct is src/esp8266_spa.ino. Thx

srasmond, 2021-04-15:
@pengemetal you could try to ping it. If it's connected you should also see it in your network list with all devices. Nornally, when using the code from this repo, the Wemos should broadcast its own SSID as well, so in the list of wifi network it should be visible as well.
Other things you could check is the serial output when connected with the board via usb.

pengemetal, 2021-04-15:
> @pengemetal you could try to ping it. If it's connected you should also see it in your network list with all devices. Nornally, when using the code from this repo, the Wemos should broadcast its own [...]

Ping do nothing. I can see an ESP-xxxxxx SSID, but I use many esp I don't know it's wemos. I tried Platformio serial monitor, but without any success:
> Executing task in folder esp8266_spa-master: C:\Users\Admin\.platformio\penv\Scripts\pio.exe device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on COM7  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
���␙
��D�������␄!>�@���H!�4�L�����?

pengemetal, 2021-04-15:
At baudrate 115200:

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

pengemetal, 2021-04-18:
I tried with a NodeMCU v3, problem the same:

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

srasmond, 2021-04-18:
@pengemetal and did you try the solution from higher in this post (disabling the reset function)?

pengemetal, 2021-04-18:
> @pengemetal and did you try the solution from higher in this post (disabling the reset function)?

I need to delete this row?
if (WiFi.status() != WL_CONNECTED) {
//hardreset();
}

Thx

srasmond, 2021-04-18:
No, you just need to comment out the hardreset function where needed. The // in the example you showed is a way of commenting a line (so you don't need to delete it and easily can activate it again afterwards).
For testing I suggest disabling the hardreset for the wifi connection, but also for the serial check (if the device is not hooked up correctly, it will keep resetting as well).

pengemetal, 2021-04-18:
> No, you just need to comment out the hardreset function where needed. The // in the example you showed is a way of commenting a line (so you don't need to delete it and easily can activate it again [...]

Notthing has changed :( messege still the same

EmmanuelLM, 2021-04-18:
Are you connected to the hot tub yet, or is that solo?

pengemetal, 2021-04-18:
> Are you connected to the hot tub yet, or is that solo?

Both

EmmanuelLM, 2021-04-19:
Looks like a potential compile/flash error as you shouldn't see any ESP-xxxxxx SSID

 Questions to help debug:
- are you using atom/platformio to compile?
- do you know the memory size of your chip?
- A guess is your chip has a lower flash size - have a look at "board_build.ldscript = eagle.flash.4m3m.ld" and change to "board_build.ldscript = eagle.flash.1m0.ld"

srasmond, 2021-04-19:
@EmmanuelLM , why shouldn't you see any SSID? There is nothing in the code that prevents the esp from broadcasting the SSID. I see it as well. 
It's a nice tip hoever: people should prevent this broadcasting or put a WPA key on the esp. Else you have a potential security issue at hand.

I do however also believe that there might be some flash error in this case like you suggested. Nice suggestion concerning the lower flash size!

pengemetal, 2021-04-20:
> Looks like a potential compile/flash error as you shouldn't see any ESP-xxxxxx SSID
>
> Questions to help debug:
>
> * are you using atom/platformio to compile?
> * do you know the memory size of your chip? [...]

Thx for your effort!

- I use platformio (Visual Studio Code win10)
- I use Wemos d1
- I flashed NodeMCU v2 and the result was the same, I think Nodemcu has different chip. I will try your suggestion, where can I change "board_build.ldscript = eagle.flash.1m0.ld"?

pengemetal, 2021-04-21:
I changed it in platformio.ini, but I get this:

Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH 'C:\Users\Admin\Downloads\esp8266_spa\esp8266_spa-master\.pio\build\d1_mini_pro C:\Users\Admin\Downloads\esp8266_spa\esp8266_spa-master\.pio\build\d1_mini_pro\ld C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\ld C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\lib C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\NONOSDK22x_190703'
Error: Could not find 'eagle.flash.1m0.ld' LD script in LDPATH 'C:\Users\Admin\Downloads\esp8266_spa\esp8266_spa-master\.pio\build\d1_mini_pro C:\Users\Admin\Downloads\esp8266_spa\esp8266_spa-master\.pio\build\d1_mini_pro\ld C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\ld C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\lib C:\Users\Admin\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lib\NONOSDK22x_190703'

srasmond, 2021-04-21:
Could you try with eagle.flash.1m.ld (so without the "0").
Also a link that could be of intrest in this case : 
https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

pengemetal, 2021-04-21:
I tried, flash is perfect, but the wdt fault is still existing :(

srasmond, 2021-04-21:
How is your esp currently powered for testing? I'm guessing you are just plugin it in via an usb adapter to the laptop, so you can check the messages in the serial monitor?

pengemetal, 2021-04-21:
Yes, laptop via usb.

EmmanuelLM, 2021-04-21:
This may not work as terminal is taken over to communicate with the hot tub.
I find your issue very odd - can you try flashing something else on yoru
chip?

pengemetal, 2021-04-22:
Yes, I flashed with esphome and worked perfect! I am thinking about using a D1 mini pro. I have one, but it is built for control sauna.

pengemetal, 2021-05-04:
I try a D1 mini pro, almost the same result, but now I can see nothing in serial monitor. I think the main problem is connecting to wifi. I can't see Wemos fixed ip in the network. When I flash esphome, there is no problem.

pengemetal, 2022-01-10:
> @pengemetal Can this be closed? Is your issue resolved and you can enjoy the sketch?

Yes, thank you! It solved, I dont know what was wrong, I think the wifi signal was the problem! Finally the sketch works great! :)

closed 2022-01-11 by cribskip