cribskip/esp8266_spa #15: OTA Update And HASS Client

qwacko, 2021-04-16:
Hi, Thanks heaps for the logic. I had been trying with Micropython to get this going, but I was just unsuccessful at getting it to go. Then just a couple of hour and I got your solution going. It would be great to get this as part of ESPHome, but I think the refresh rate may be too low.

Anyway, I have made a couple of changes to my version of the code to make it work how I wanted, and thought you may appreciate these minor code changes.

**Arduino OTA**
I see you had a note to get OTA to work (and even had the library included). The logic changes to make this are:

Just incude ArduinoOTA.begin(); in the setup task.
`void setup() {
   ...
   ArduinoOTA.begin();
   ...
}
`

And include the ArduinoOTA.handle at the start of the loop:
`void loop() {
  ArduinoOTA.handle();...`

And it will be ready. Once you have done this intial serial download, then you can add the following lines to the platformio.ini file:
```
upload_protocol = espota
upload_port = spa.local
```

And through the magic of mDNS it will be able to do a download to it. This is great for debugging while it is connected to the SPA.

**HASS Client**
I noted that the HASS client was only working for some items. Most markedly the Climate item wasn't working, this is due to the default message length limit of 256B. You may have fixed this by modifying the PubSubClient.h file in your instance (or maybe the climate just hasn't worked for you). To make it work for me, I added the "setBufferSize" comand when the mqtt service is setup.

```
  mqtt.setServer(BROKER.c_str(), 1883);
  mqtt.setCallback(callback);
  mqtt.setKeepAlive(10);
  mqtt.setSocketTimeout(20);
  mqtt.setBufferSize(1024); // Set buffer size to allow discovery messages to work correctly.
```

EmmanuelLM, 2021-04-19:
Sweet - please could you send a pull request vs 0.36 (c.f. my branch)?

also see line 488 for setBufferSize - I had increased it to 512, was that not enough?

flechaig, 2021-12-22:
In my experience with a Balboa BP6013G2, I had to set : 
`mqtt.setBufferSize(1024);`
for everything to work as expected.

Gamerayers, 2022-04-19:
Is there any reason that it couldn't be ESPHome?  Are you saying the refresh rate of the ESPHome software runs to slow to grab it off of modbus or simply that the data would be too slow to Home Assistant.  It isn't like you are sending data and needing it in a time critical nature.  Even a couple seconds of delay isn't a big deal.