cribskip/esp8266_spa #16: Problem finding the right pins on the pcb

Warrior700, 2021-04-20:
Hello everybody,

I have a Balboa bp2100g3 board. Unfortunately, I'm not really sure which pins the RS485 transceiver A / B should be connected to. I have a picture of the circuit board, the connector is on the top right. I've found the physical link layer description and I am confused what the meaning is. "It is also recommended to put pin 3 on the return line of the connection device to reduce the common mode voltage." Is It really necessary?

How did you connect your transceiver? A schematic would be helpful.

Many thanks to all the developers of this project.

[image]

cribskip, 2021-04-20:
Nice picture :-)

Are you able to measure the voltages on the pins? One pair should show +11...15V. So you know GND and Supply.

The other pair shows fluctating low voltages (<3V) to ground. The higher is A, IIRC. Those are interchangeable if it does not work, don't worry. The other two pins are NOT interchangable!!!!

srasmond, 2021-04-21:
@Warrior700 I have the same board layout. I kept it really simple as well:
- I plugged in an ATX Molex Micro Fit Connector 4Pin cable (60cm) in to the J35 plug on the Balboa board (the 2nd 4 pin connector from the top on the right of your picture).
- On the other end of the Molex cable (where I cut off the other connector to have 4 loose wires) you usually have 2 black wires and a yellow and red wire. 
- In my case, the 2 black wires where the data connection (A & B), yellow was GND and Red was my power supply.
- **This could differ for you! You definitely need to measure this!**
- I used the Earth cable of my spa to connect my meter's ground with a clamp and then checked DC voltages on each wire.
- GND (0V) and V (+/- 13V in my case) are the easiest to detect and can be attached to your step-down converter.
- A & B are lower voltage and close to each other, but you should see a tiny difference, with the higher VDC one being cable B ("_RS-485 B"_) (cable A should be little lower in VDC, but like cribskip mentioned, no worries if they are connected wrongly, you will get bad messages, but no electrical damage or what so ever, just switch them if necessary on your TTL To RS485 Adapter --> **but like it was mentioned by cribskip always leave GND and V alone!**). 
- That's it. No other tricks or cables where used to connect the Balboa board to my custom board.

Warrior700, 2021-04-21:
Thanks for the detailed description, I have a Molex Micro Fit cable like you in stock, also with black, yellow and red. At the weekend I will measure the pins. At first I will only connect RX / TX. For testing, I can also connect the WEMOS via USB and supply the 3.3V + GND via breadboard to the TTL to RS485 pcb. This should be possible, right?

srasmond, 2021-04-21:
For flashing and testing if the ESP can be found on your network and communication is working I just connect the _WEMOS_ via USB.

For testing the Spa communication without supplying power from the Balboa board, yes, you could power your _TTL to RS485_ via the 3.3V and GND pins from the _WEMOS_ if its connected via USB.

If that is working, ideally, before moving on, I would also test it without powering the _WEMOS_ via USB, but for example by wiring everything up and connecting an other 5V power source, so you could test the output of the step-down converter (which should be 3.3V) and receive your MQTT messages.
I made my PCB so that I could easily plug in and out both the _TTL to RS485_ and _WEMOS_ boards.
That way I could also double check my output voltage from the converter when it's straightly connected to the Balboa board before endangering my components on the PCB.

Michael0yodi, 2021-05-19:
What are the expected voltages? I've read that it should be ~3volt on RS485 but this is my readings, am i good or do you think somethings is wrong?
[image]
[image]

cribskip, 2021-05-19:
Perfect, go for it. You identified supply and gnd. If you mix up rs485 a
and b just swap them

flechaig, 2021-12-22:
In my experience, with a Balboa BP613G2, the pinout is exactly the same.