cribskip/esp8266_spa #17: no connection, no messages

pato-loco, 2021-06-09:
Hello everybody,

I am desperate about this project right now.

My config:
- BP2100 (Spatech)
- WEMOS D1 mini pro (including external antenna)
- ARCELI TTL To RS485 Adapter 485 Serial Port UART Level Converter Module 3.3V 5V
- LM2596 (set to 3.3V)

What happened until now:
- Adoption of version 0.34
- Adaptation to my environment (SSID, PWD, Broker, Brk_Login, Brk_pwd)
- no connection to the WLAN recognizable

Then change by me:
- //Serial.println("millis-Hrest ");
- // hardreset ();
- MQTT client send a message to MQTT server if connected

Result:
- no hard reset
=> WLAN probably has enough time to connect and the MQTT server receives a "connection" confirmation
- that's all

Problem:
It doesn't matter whether I connect the spa or not
(RX flashes when connected), I do not receive any messages via MQTT.
When entering http://my_ip/update I only get the option for OTA update. No login process. he default website then shows "Not found: / config".

I am happy about any kind of support.

Thanks in advance!

pato-loco, 2021-06-09:
Hi Sascha,

thank you for your fast response. And thank you for your great work :-)

Mhh, from my sight of view I have done all things, what is wright down in the readme.
I´ve also crossed A and B from the Spa. No connection to Wifi (without passing the hardreset).

The Wifi-Client connects to Wifi if I commend out "//Serial.println("millis-Hrest ");"
So I think, the Wifi range is sufficient.

In one of the "Issue-comments" I read about, that I have to bring "all" GND to one point. What does it mean?
Should I also bring GND (-) from Spa-In (-) to a common point?

And is there a chance to "debug" the input signal from the spa to the serial console or to mqtt?

pato-loco, 2021-06-09:
[image]
[image]
[image]

pato-loco, 2021-06-09:
// https://github.com/ccutrer/balboa_worldwide_app/blob/master/doc/protocol.md
// Reference:https://github.com/ccutrer/balboa_worldwide_app/wiki

// Please install the needed dependencies:
// CircularBuffer
// PubSubClient

// TODO:
// HomeAssistant autodiscover - mostly done
// Configuration handling
// Proper states (rather than just ON/OFF)
// OTA update from Firebase
// Settings
// Implement Existing Client Request/Response
// Handle non-CTS Client IDs

// +12V RED
// GND  BLACK
// A    YELLOW
// B    WHITE
#include <LittleFS.h>
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <CircularBuffer.h>
#include <ESP8266WebServer.h>   // Local WebServer used to serve the configuration portal
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ArduinoOTA.h>
#include <PubSubClient.h>       // MQTT client

#define VERSION "0.35_1"
const char* HOSTNAME = "hottube";
String WIFI_SSID = ":-)";
String WIFI_PASSWORD = ":-)";
String BROKER = "192.168.9.44";
String BROKER_LOGIN = "foo";
String BROKER_PASS = "bar";
#define AUTO_TX false //if your chip needs to pull D1 high/low set this to false
#define SAVE_CONN false //save the ip details above to local filesystem

#define STRON String("ON").c_str()
#define STROFF String("OFF").c_str()

//HomeAssistant autodiscover
#define HASSIO true

#define TX485 D1  //find a way to skip this
#define RLY1  D7
#define RLY2  D8

CircularBuffer<uint8_t, 35> Q_in;
CircularBuffer<uint8_t, 35> Q_out;

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

WiFiClient wifiClient;
PubSubClient mqtt(wifiClient);

extern uint8_t crc8();
extern void ID_request();
extern void ID_ack();
extern void rs485_send();

uint8_t x, i, j;

uint8_t last_state_crc = 0x00;
uint8_t send = 0x00;
uint8_t settemp = 0x00;
uint8_t id = 0x00;
[... 725 lines cut ...]
    } else if (Q_in[2] == id && Q_in[4] == 0x28) {
      if (last_state_crc != Q_in[Q_in[1]]) {
        decodeFault();
      }
    } else if (Q_in[2] == 0xFF && Q_in[4] == 0x13) { // FF AF 13:Status Update - Packet index offset 5
      if (last_state_crc != Q_in[Q_in[1]]) {
        decodeState();
      }
    } else {
      // DEBUG for finding meaning
      //if (Q_in[2] & 0xFE || Q_in[2] == id)
      //print_msg(Q_in);
    }

    // Clean up queue
    _yield();
    Q_in.clear();
  }

  // Long time no receive
  if (millis() - lastrx > 5000) {
    //Serial.println("millis-Hrest");
    //hardreset();
  }
}

pato-loco, 2021-06-15:
"put all GND together was the solution" top! THX you!

closed 2021-06-15 by pato-loco

Sch-LikA, 2022-11-07:
Hi, I seem to have the same issue (using a esp8266 "module"), could you please be a bit more specific ? What ground was missing ? Should I connect the third connection on the RS485 (the Chinese character on the right of A+ B- in the last photo) to ground ?

Thanks !