cribskip/esp8266_spa #18: Strange Temperature Readings in HASS

reicharb, 2021-06-10:
My install seems to be working and I am getting data into HASS.  I see a couple issues maybe someone can help me with.
1. Strange Temp readings.  example: Spa says 92 deg F, HASS is showing 46 F.  I am sure it has to do with some kind of conversion.
2. Pump 1...My spa has a 2 speed pump.  During low speed HASS shows the pump off and it only shows on when the pump is in high speed.
3. During HASS reset, it show unavailable.  I have to cycle power on the 8266 and then the reading come in.

EmmanuelLM, 2021-07-08:
Hello.... let me have a look at issues 1 & 2 for you, I don't think any of us have a Farenheit reading nor 2 speed pump so this is a good case study :-)

On #1, I see that we divide the temperature in 2 for the Celcius, I will check where the Farenheit flag is
On #2, high speed is not (yet) supported - I will add that at the weekend, it seems easy enough, to help me debug, could you please send me the mqtt output for "Spa/config/*"?
On #3, I have the same annoyance, leave it with me

EmmanuelLM, 2021-07-10:
hmmm, second problem is a tad more problematic as high speed toggle is not documented... will have to dig into other people's code.
3 may be fixed (untested in my branch)

reicharb, 2021-07-10:
I was able to figure out #1, I did find the math function and made the change, it now reads Fahrenheit.
I also figured out #2, I created a low speed pump status.  
#3 still seems to be a problem, a little work around I created a HASS automation that drops the 8266 from the Unifi wifi and waits for 2 minutes to allow it to connect to the network again.  I believe it forces it to hard reset itself.

EmmanuelLM, 2021-07-10:
Amazing, would you mind sending a pull request against my branch?

EmmanuelLM, 2021-07-11:
Hey, could you try v0.37.3 from here: https://github.com/EmmanuelLM/esp8266_spa
This has a fix for your first problem, I'll work on the second one - I am curious how you fixed it? I was thinking about using the new 'Select' configurator from Home-Assistant

EmmanuelLM, 2021-07-11:
Issue 3 should be fixed as well

Gamerayers, 2022-04-22:
I believe I have similar issues as this.  I'm super happy that I'm finally able to communicate with the spa using the ESP board, but things aren't working exactly as I expect, and I believe it is a different in models.
1. As you can see from the screenshot and mqtt messages, my spa is set for 115 and is not heating, but the image of the control panel says my spa is heating and set for 93 with a current temp of 92.
2. I have 2 control buttons for jets (the Jet button and the Aux button).  The jet button allows for 2 speeds, but the controls for hot tub jet1 and jet2 don't seem to allow for that control actually.
3. Heating mode was on/high range on when I first turn on the spa, but if I turn if off from the Home assistant, it does nothing and doesn't come back on when I turn it on again from Home Assistant.  I have no ability to really control this from the control panel.
The LED lights sort of control properly and do register when I turn them on from the control panel.

I'm guessing this is all due to my spa version/control panel version, but not sure how to make adjustments to make it work correctly.

`{"name":"Hot tub heating mode","uniq_id":"ESP82Spa_3","cmd_t":"Spa/heatingmode/set","stat_t":"Spa/heatingmode/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub high range","uniq_id":"ESP82Spa_4","cmd_t":"Spa/highrange/set","stat_t":"Spa/highrange/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub light","uniq_id":"ESP82Spa_7","cmd_t":"Spa/light/set","stat_t":"Spa/light/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub jet1","uniq_id":"ESP82Spa_8","cmd_t":"Spa/jet_1/set","stat_t":"Spa/jet_1/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub jet2","uniq_id":"ESP82Spa_9","cmd_t":"Spa/jet_2/set","stat_t":"Spa/jet_2/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Filter 1 start","val_tpl": "{{value_json.start}}","uniq_id":"ESP82Spa_11","stat_t":"Spa/filter1/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Filter 2 start","val_tpl": "{{value_json.start}}","uniq_id":"ESP82Spa_12","stat_t":"Spa/filter2/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Filter 1 duration","val_tpl": "{{value_json.duration}}","uniq_id":"ESP82Spa_13","stat_t":"Spa/filter1/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Filter 2 duration","val_tpl": "{{value_json.duration}}","uniq_id":"ESP82Spa_14","stat_t":"Spa/filter2/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub status","uniq_id":"ESP82Spa_1","stat_t":"Spa/node/state","platform":"mqtt","dev":{"ids":["ESP82Spa"],"name":"Esp Spa","sw":"0.37.4"}}
{"name":"Hot tub heating state","uniq_id":"ESP82Spa_6","stat_t":"Spa/heatstate/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Filter 2 enabled","uniq_id":"ESP82Spa_15","stat_t":"Spa/filter2_enabled/state","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}
{"name":"Hot tub thermostat","uniq_id":"ESP82Spa_0","temp_cmd_t":"Spa/target_temp/set","mode_cmd_t":"Spa/heat_mode/set","mode_stat_t":"Spa/heat_mode/state","temp_unit": "C","curr_temp_t":"Spa/temperature/state","temp_stat_t":"Spa/target_temp/state","min_temp":"27","max_temp":"40","modes":["off", "heat"], "temp_step":"0.5","platform":"mqtt","dev":{"ids":["ESP82Spa"]}}`

[image]
[image]