cribskip/esp8266_spa #26: WEMOS D1 Mini Pro clone - not working

mopac, 2022-04-18:
Seems the genuine D1 mini pro is in short supply, so I had to buy a clone!
The problem is the clone uses CH340 for the USB to Serial whereas the original uses the CP2104. It also has no series resistor in the Rx path.
CH340 when the USB is not connected drives the RX pin low. This blocks all serial traffic from the hottub.

The solution I came up with was to use SoftwareSerial on pin 4 & 5, connected to the RS485 converted.
Running Software Serial at 115200 baud is at its limit, but it works with only a few errors every minute. These are normally quickly overwritten by correct data.
Maybe I'll add better data checking so these errors don't get passed to MQTT.

Has anyone else had problems with clones?
Any other the solutions than SoftwareSerial?

Ray

Gamerayers, 2022-04-19:
How were you able to find out that software serial was the fix.
I've been trying to figure out why my device isn't getting connected to MQTT, but I know that it needs to communicate to the spa first (btw, this is ridiculous for troubleshooting issues).
I see it connect to the wifi briefly, and I also see the blue LED flash, but nothing else.
This is the D1 Mini Pro I have.
https://www.amazon.com/gp/product/B08HH96RJY/ref=ppx_yo_dt_b_asin_title_o06_s00?ie=UTF8&psc=1
This is the adapter I have for communicating with RS485.
https://www.amazon.com/gp/product/B08XLT21S6/ref=ppx_yo_dt_b_asin_title_o09_s01?ie=UTF8&psc=1

Any help would be greatly appreciated.  I have a step down converter to 3.3V and it works.  And so far, no issues where the screen is blanks, so I assume the pins are connected correctly, but no data.

Gamerayers, 2022-04-19:
Just a quick update, it is definitely the communication with the spa over rs485 that is the issue, but the problem is how to determine if its the D1 Mini Pro clone that is the issue.  I'm able to see it connect and disconnect from my mqtt server, so it isn't the wifi or the mqtt, but it never sends anything.
So what changes to the code are needed to make this work/what other things do I need to do to accomplish the change.

Honestly, the errors every couple minutes are a pain, but if I can at least get that far, maybe I can get a genuine D1 at some point to fix it.

mopac, 2022-04-19:
Line 48 in the code
#define PRODUCTION true
change that to 
#define PRODUCTION false
Then you don't get resets about every 30 seconds. So it should stay connected to MQTT broker.
Then publish debug info to MQTT and see what it is doing.
Or connect the USB cable to your laptop and publish debug info to Serial.
Line 867 is the start of the loop to get serial bytes from the hottub. You can add a print statement in there to see what is arriving.
It should be a long series of bytes starting and ending with 0x7E - maybe 25 bytes long.

My problem was the Rx pin was permanently low when no USB was attached; but was wiggling when it was attached.

mopac, 2022-04-20:
I came up with an even better solution that I think fixes most of the issues with getting these boards to work!
The fundamental problem is that ESP8266 only has one usable UART and that must be shared between USB serial and RS485 serial to the hottub.
But the ESP8266 has the ability to route the UART away from pins 1 & 3 (where the USB to Serial is connected), to pins 13 & 15.
My board has the RS485 converter connected to pins 13 & 15 and the WEMOS D1 Mini Pro has the USB to Serial connected to pins 1 & 3.
So now the the two sets of serial pins are completely separate.
So I switch the hardware UART to pins 13 & 15 and  use SoftwareSerial library for the USB on pins 1 & 3
Code:
  // Using hardware Serial for RS485 and software Serial for USB
  Serial.begin(115200);
  Serial.swap(); // Switch Hardware Serial to GPIO13 & GPIO15

  swSer1.begin(115200,SWSERIAL_8N1, 3, 1, false); // Start software serial
  swSer1.enableIntTx(false); // This is needed with high baudrates

  if (!swSer1) { // If the object did not initialize, then its configuration is invalid
      Serial.swap(); //Switch the Serial back to 1 & 3
      Serial.println("");
      Serial.println("Invalid SoftwareSerial pin configuration, check config"); 
  }
  else {
      swSer1.println("");
     swSer1.println("Correct SoftwareSerial pin configuration. Using Software Serial for USB");   
  }

Then all the Serial.print statements in the rest of the code need to use swSer1.print.

Working really well with no errors from the Hottub and no obvious errors in the debug printing

Gamerayers, 2022-04-22:
So after much troubleshooting, and wasted time, the issue is simply that the designer of the RS485 board is not an electrical engineer, and doesn't practice standards.  And I guess I can't read.
Who makes black power and red ground.
[image]

sdrshnptl, 2024-02-20:
> So after much troubleshooting, and wasted time, the issue is simply that the designer of the RS485 board is not an electrical engineer, and doesn't practice standards.  And I guess I can't read.
> Who makes black power and red ground.
> [image]
> 

[image]

Thank you!
First thing I did this!

cribskip, 2024-02-20:
Thank you for posting the solution to this. Is your SPA connection now working?