cribskip/esp8266_spa #28: Helping editing code so that I can simply public data to MQTT

Gamerayers, 2022-04-25:
After I finally got it hooked up to my hot tub on Friday, I've been trying to figure out how to understand why the data is all incorrect.
The temperature is really bad, but ultimately, I want to publish the entire read data, all the flags, from the spa to 1 or 2 MQTT messages.  What's the easiest way to do this.
I'm guessing, based on reading the information on the wiki, that my hot tub is using differen bits for most things on here.  My Jet has 2 speeds, but the jets on here is only 1.  My aux controls a jet, but there are 2 controls for jets on here which are different.  My thinking is go through, figure out why the data is all off.  Once I have the correct data, I can realign the signals and publish it correctly.

I guess where I'm most confused at this point is this line of code:   if (bitRead(Q_in[18], 1) == 1).
Why is Q_in[18] byte 13?

It seems like both the developers working on this maybe aren't working on anything on github since Oct 2021.  It would be helpful is anybody has some insight on to how to make this work for my specific spa.  I know I'm really close, but getting the exact right parameters maybe is the hardest part.  I am concerned that the temperature is off by so much though.

cribskip, 2022-04-25:
Hey,

thank you for opening the issue. I've not neglegted the project - rather, I'm glad to enjoy it working ;-)

So your tub is completely different than mine in its configuration. I think we can work this out. Please post as many details of the controller you can get hold of (control unit, controller...) for others to find this quicker.

Your first step should be to remove the comment slashes in this line:
https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/esp8266_spa.ino#L865

Then you get on Spa/rcv the received data stream. Then we can check if the messages are the same format.

Additionally, you referenced 
https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/esp8266_spa.ino#L418

Those comments before each block helped me to keep in sync with the wiki. It simply means: Q_in[18] holds byte 13 of the wiki documented message. Wiki documentation simply used an offset to my implementation and I didn't want to drop bytes just to make it fit.

HTH
Sascha

Gamerayers, 2022-04-26:
Hi Sascha,

Thanks for the reply.  I've made some progress, but definitely some odd things.  I had to swap out the board since the previous one must have been shocked or something and can't be updated by USB and stopped communicated.  No biggy because the different board is acting a little different.
Looking at the below, it appears that the temp is correct, but maybe is being scaled to F even though it is already in F.  93 is what my spa is set to in F, but Home Assistant now says it is 199.  The Jet_1 and Jet_2 appear to be correct now.  Light is also correct, but when I try to command them from the Home Assistant, it doesn't do anything.
RCV goes through a number of values, but it moves too quickly.  I've posted a graph below.
Highrange, heat state, heat_mode, blow and circ aren't really options on my spa, so might be part of it.  I also can't enable heatingmode on the hot tub, so not sure if this is something simply that should be a sensor instead of a switch for my tub.
[image]
[image]
[image]

Info about my hot tub.  Its a 2020 Master Spa.  Clarity Series Balance 7.  Here's some images.
[image]
[image]
[image]

cribskip, 2022-04-26:
Spa/rcv contains the instantly received bytes and is very fast for human eye :-)
Please consider viewing it with a MQTT Client capable of displaying the history (e.g. mosquitto-sub on Linux).

Really seems to do C to F conversion twice. Maybe you're the first user with imperial units. Welcome :-D We may fix this with a compile time define-switch for now. 

Did you see Spa/config messages? If now, this may cause the conversation problem:
https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/esp8266_spa.ino#L307

I'd really like to view one or two complete messages from your tub using the rcv message channel. Just to get to know your controller better and decide on the way to go.

Cheers,
Sascha

Gamerayers, 2022-04-26:
I don't seem to see the config message.  Simply doesn't ever seem to populate.  So I'm guessing that is part of the issue with the temperature. (image below)

As the messages doesn't seem to repeat exactly, I am not sure where the start of the rcv message and the end is.  So I logged it for a little bit.
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/brian/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/brian/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Topic: Spa/rcvQoS:   0 | 17 | 2022-04-26 13:55:45:277
-- | -- | --
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:277
Topic: Spa/rcvQoS: 0 | 62 | 2022-04-26 13:55:45:278
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:278
Topic: Spa/rcvQoS: 0 | 5 | 2022-04-26 13:55:45:278
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:278
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:279
Topic: Spa/rcvQoS: 0 | 6 | 2022-04-26 13:55:45:279
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:279
Topic: Spa/rcvQoS: 0 | 92 | 2022-04-26 13:55:45:279
Topic: Spa/rcvQoS: 0 | 7 | 2022-04-26 13:55:45:280
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:280
Topic: Spa/rcvQoS: 0 | 17 | 2022-04-26 13:55:45:280
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:280
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:280
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:281
Topic: Spa/rcvQoS: 0 | 62 | 2022-04-26 13:55:45:281
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:281
Topic: Spa/rcvQoS: 0 | 5 | 2022-04-26 13:55:45:282
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:282
Topic: Spa/rcvQoS: 0 | 6 | 2022-04-26 13:55:45:282
Topic: Spa/rcvQoS: 0 | 92 | 2022-04-26 13:55:45:282
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:282
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:283
Topic: Spa/rcvQoS: 0 | 7 | 2022-04-26 13:55:45:283
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:283
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:283
Topic: Spa/rcvQoS: 0 | 17 | 2022-04-26 13:55:45:284
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:284
Topic: Spa/rcvQoS: 0 | 62 | 2022-04-26 13:55:45:284
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:285
Topic: Spa/rcvQoS: 0 | 5 | 2022-04-26 13:55:45:285
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:285
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:285
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:286
Topic: Spa/rcvQoS: 0 | 7 | 2022-04-26 13:55:45:286
Topic: Spa/rcvQoS: 0 | 92 | 2022-04-26 13:55:45:286
Topic: Spa/rcvQoS: 0 | 6 | 2022-04-26 13:55:45:286
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:287
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:287
Topic: Spa/rcvQoS: 0 | 17 | 2022-04-26 13:55:45:287
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:287
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:288
Topic: Spa/rcvQoS: 0 | 5 | 2022-04-26 13:55:45:288
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:288
Topic: Spa/rcvQoS: 0 | 62 | 2022-04-26 13:55:45:288
[... 103 lines cut ...]
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:324
Topic: Spa/rcvQoS: 0 | 7 | 2022-04-26 13:55:45:324
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:324
Topic: Spa/rcvQoS: 0 | 0 | 2022-04-26 13:55:45:325
Topic: Spa/rcvQoS: 0 | 17 | 2022-04-26 13:55:45:325
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:325
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:326
Topic: Spa/rcvQoS: 0 | 5 | 2022-04-26 13:55:45:326
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:326
Topic: Spa/rcvQoS: 0 | 62 | 2022-04-26 13:55:45:326
Topic: Spa/rcvQoS: 0 | 6 | 2022-04-26 13:55:45:327
Topic: Spa/rcvQoS: 0 | 92 | 2022-04-26 13:55:45:327
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:327
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:327
Topic: Spa/rcvQoS: 0 | 126 | 2022-04-26 13:55:45:328
Topic: Spa/rcvQoS: 0 | 16 | 2022-04-26 13:55:45:328
Topic: Spa/rcvQoS: 0 | 191 | 2022-04-26 13:55:45:328
Topic: Spa/rcvQoS: 0 | 7 | 2022-04-26 13:55:45:328
Topic: Spa/rcvQoS: 0 | 17 | 2022-04-26 13:55:45:329

</body>

</html>

[image]

cribskip, 2022-04-26:
Those "126" markers are the begin and end of transmission bytes (0x7E) see
the wiki for further information, I'm sure now you can spot the telegrams

Gamerayers, 2022-04-26:
Ah, thats much better....  Just have to read a little better.
The messages are still a little garbled at times and I'm guessing that is just the order that it comes in and the speed that it is being recorded in.
I have tried to make the best of it, and made the table, but also instead the excel data I have in its more raw format.
[mqtt data.xlsx](https://github.com/cribskip/esp8266_spa/files/8566248/mqtt.data.xlsx)

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

[mqtt data.xlsx](https://github.com/cribskip/esp8266_spa/files/8566244/mqtt.data.xlsx)

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/brian/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/brian/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

7E | 5C | 7 | 11 | 0 | 10 | BF | 3E | 7E
-- | -- | -- | -- | -- | -- | -- | -- | --
7E | 10 | 7 | BF | 11 | 0 | 0 | 3E | 7E
7E | 7 | 10 | 11 | BF | 0 | 0 | 3E | 7E
7E | 5 | 10 | 6 | 5C | BF | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | BF | 6 | 5C | 10 | 7E |   |  
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 5C | 7E | 6 |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 0 | 0 | 11 | 3E | 7E
7E | 5 | BF | 6 | 10 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 5 | BF | 6 | 5C | 7E | 10 |   |  
7E | 7 | 10 | BF | 0 | 3E | 7E | 0 | 11
7E | 5 | 10 | BF | 6 | 7E | 5C |   |  
7 | 7E | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 7E | 5C |   |  
7E | 7 | 10 | 11 | 0 | BF | 0 | 3E | 7E
7E | 5 | BF | 6 | 5C | 10 | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 7 | 5C | 6 | 10 | BF | 11 | 0 | 0
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 5C | 6 | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | BF | 6 | 5C | 7E | 10 |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | BF | 6 | 5C | 10 | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
7E | 5 | 10 | BF | 6 | 5C | 7E |   |  
7E | 7 | 10 | BF | 11 | 0 | 0 | 3E | 7E
5 | 7E | BF | 6 | 5C | 10 | 7E |   |  
7E | 5 | 10 | 3E | 6 | 5C | 7E |   |  

</body>

</html>

cribskip, 2022-04-27:
Looking good but I miss state updates to adress 0xFF (e.g. 7E xx FF ....).
Those messages contain the state of the tub and the data you're looking for

https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/esp8266_spa.ino#L962

Gamerayers, 2022-04-28:
How often does that message send?  I logged for a couple minutes and can't find an FF in it at all.  There's over 100,000 hex pair sent, so not sure how much I should log to find this.  Is it only sent every couple minutes? hour? when I turn the hot tub on?
Or should I be logging something other than Spa/rcv?

Gamerayers, 2022-04-29:
Well, I figured out a lot of information over the course of this and made some cool formulas in Excel.  Attached is the messages over about 12 minutes or so, but specifically this:

7E20FFAF1300005D062A00005D0000000000000000000000005D000000780000937E
7E20FFAF1300005D062B00005D0000000000000000000000005D0000007800002A7E
7E20FFAF1300005D062C00005D0000000000000000000000005D0000007800000C7E
7E20FFAF1300005D062D00005D0000000000000000000000005D000000780000B57E
7E20FFAF1300005D062E00005D0000000000000000000000005D000000780000797E
7E20FFAF1300005D062F00005D0000000000000000000000005D000000780000C07E
7E20FFAF1300005D063000005D0000000000000000000000005D000000780000947E
7E20FFAF1300005D063100005D0000000000000000000000005D0000007800002D7E
7E20FFAF1300005D063200005D0000000000000000000000005D000000780000E17E
7E20FFAF1300005D063300005D0000000000000000000000005D000000780000587E
7E20FFAF1300005D063500005D0000000000000000000000005D000000780000C77E
7E20FFAF1300005D063600005D0000000000000000000000005D0000007800000B7E

is what we are looking for.  I removed the duplicate messages and anything that was mis formatted (dropped byte, bit, etc.).

[mqtt data3.xlsx](https://github.com/cribskip/esp8266_spa/files/8590790/mqtt.data3.xlsx)

Gamerayers, 2022-04-29:
Well, since I couldn't find the config messages, I actually enabled some more of the debug code you have, and low and behold, my spa gives me a 1 for temperaturem but is using F, so the scaling is all off.

[image]

Gamerayers, 2022-05-02:
K, after much debugging, I switched the temperature configuration and everything basically functions now.

I saw basically because I set the hot tub to 99 from 93 last night, it warmed up, but a minute later was like, yay, maybe I only want 98.  So I tried to change it and it kept going back to 99.  So I changed on the control panel when we got it, and it set, but then when we were done, we set it back to 93.  All looked good, but no matter what was set on the control panel, when I try to set it from Home Assistant, no matter what value I put it, it goes back to 99.  Not entirely sure why.  It seems to set, stays set for a couple seconds, temp value goes from red to white.  Then a couple seconds later, it goes back to 99 as the set point.

Gamerayers, 2022-05-08:
Would really like to say mark this item as closed, but I continue to only be able to change the temp in Home Assistant and it always sets it to 99.  No matter what I do.  i can change it on the spa, and everything seems okay.  Any idea why this would happen? @cribskip

cribskip, 2022-05-08:
I'm suspecting Home Assistant does not only send the new setpoint but some more data and parsing fails.
What does Spa/node/debug contain when you use HA?

If this line fails, results are currently not predictive to me (might be 0)
https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/esp8266_spa.ino#L641

You could also uncomment this line to see whats actually sent to the tub
https://github.com/cribskip/esp8266_spa/blob/015e468864a35a08b53799d69ec0e647002d2343/src/balboa_helper.ino#L75

Good luck - I'm curious about the outcomes :-)

Gamerayers, 2022-05-09:
Spa/target_temp/set  This is what Spa/node/debug contains.  Should it include a temperature value also?  Or is that correct?
I do see the target temp set = 96.0, but state then goes to 99.0.

I haven;t uncommented the debug line since I don't have direct access to the ESP so usb/serial debug messages are a little harder for me to obtain.  Due to the cover lifter on the spa, taking off the side panel is a real chore.

cribskip, 2022-05-10:
That's correct for the debug statement.

I got you covered - there's OTA Update built in as opening the spa really
is no fun. Use http://<ip of esp8266>/update

closed 2022-05-13 by Gamerayers