cribskip/esp8266_spa #31: Network dropping

Michael0yodi, 2022-10-26:
Hi,

I've been using this since some time, but not without issues, after a cold boot it can run for days or even weeks, and suddenly stops working. I can still see it in Unifi -its connected to an AP, I can see that the DHCP server is constantly offering an IP.

Web server not working and no ping response. I've replaced the D1 Mini with another model but issue persisted, any ideas or have anyone else seen anything similar?
Compiled using Arduino 1.8.15 so maybe I should try and update some libraries?

<html>
<body>

2022-10-26T15:48:02 | Informational | dhcpd | Internet Systems Consortium DHCP Server 1 |  
-- | -- | -- | -- | --
2022-10-26T15:47:58 | Informational | dhcpd | DHCPOFFER on 172.16.0.147 to c4:5b:be:54:7f:xx via em0 |  
2022-10-26T15:47:58 | Informational | dhcpd | DHCPDISCOVER from c4:5b:be:54:7f:xx via em0 |  
2022-10-26T15:47:54 | Informational | dhcpd | DHCPOFFER on 172.16.0.147 to c4:5b:be:54:7f:xx via em0 |  
2022-10-26T15:47:54 | Informational | dhcpd | DHCPDISCOVER from c4:5b:be:54:7f:xx via em0 |  
2022-10-26T15:47:52 | Informational | dhcpd | DHCPOFFER on 172.16.0.147 to c4:5b:be:54:7f:xx via em0

</body>
</html>

cribskip, 2022-10-26:
Sounds to me like the connection to the spa is instable. Could you please
post a picture of your setup?

Michael0yodi, 2022-10-28:
Its just connected mainly using jumper cablers, but should be fine, I've many ESP's like this, I will try and replace the DC-DC converter also even if its been stable at 3v3 when I've checked :)

[image]

Schubi1, 2022-11-28:
Hi,
i think i have the same problem as you.
It works up to many weeks an then suddenly it doesn t respond.
Curious ist that if ported it to run on an esp32. I thought that is a mistake made by me.
I am also using a unifi controller. how do you get the information aubout the DHCP requests on the unifi controller?
Or did you already find a solution to fix the problem?

Michael0yodi, 2022-11-28:
Hi,

Sorry for not following up on this,

I replaced by dc-dc converter and also changed some wlan settings, disabled BSS Transmission and PMF, if I remember correctly, I had most options disabled from before like minimum data control/fast roaming,  etc. (other esp:s were fine even with them on). This helped me and its been working fine since, but not sure what made the difference or if I was a combination of things

Schubi1, 2022-11-28:
Thanks for that information - i will go testing …