cribskip/esp8266_spa #32: Temp going haywire!

davidmpye, 2023-03-07:
My Spa has been plodding along happily for a week or so, then it seems a confused message led to the temperature thinking it was 70'C where it has persisted...

The MQTT debug output looks like this:

Spa/time/state 07:16
Spa/heatingmode/state ON
Spa/heat_mode/state heat
Spa/heatstate/state OFF
Spa/highrange/state ON
Spa/jet_1/state OFF
Spa/jet_2/state OFF
Spa/circ/state OFF
Spa/blower/state OFF
Spa/light/state OFF
Spa/relay_1/state OFF
Spa/relay_2/state ON
Spa/target_temp/state 35.00
Spa/time/state 07:17
Spa/heatingmode/state ON
Spa/heat_mode/state heat
Spa/heatstate/state OFF
Spa/highrange/state ON
Spa/jet_1/state OFF
Spa/jet_2/state OFF
Spa/circ/state OFF
Spa/blower/state OFF
Spa/light/state OFF
Spa/relay_1/state OFF
Spa/relay_2/state ON
**Spa/node/msg 7e 05 fe bf 00 ac 7e 
Spa/node/msg 7e 08 fe bf 02 23 f1 73 ab 7e 
Spa/node/id 35**
Spa/target_temp/state 70.00
Spa/time/state 07:18
Spa/heatingmode/state ON
Spa/heat_mode/state heat
Spa/heatstate/state OFF
Spa/highrange/state ON
Spa/jet_1/state OFF
Spa/jet_2/state OFF
Spa/circ/state OFF
Spa/blower/state OFF
Spa/light/state OFF
Spa/relay_1/state OFF
Spa/relay_2/state ON
Spa/target_temp/state 70.00

Not quite sure what has happened, would welcome any suggestions :-)

David

cribskip, 2023-03-07:
Please post a bunch of msg debugs to see if the value for the temp changes.

Also, which rs485 module do you use?

davidmpye, 2023-03-10:
Hi

I have had a few more issues of similar problems and one where it crashed the controller, so I am suspecting faulty hardware (likely the RS485 board)

WIll change it and see if that fixes the problem.

I was using this one:

[image]

David

cribskip, 2023-03-11:
Hey David,

I'm using the same and also see increasing transmission errors in the last
days. On another project one of these got hot to the point where the chip
self destroyed. The quality seems to be fluctuating on these modules.

Anyway, I also suspect the rs485 setup. Could you please do me a favor and
measure the resistance between the two data lines between each other and
ground while the entire pool is shut off? Please disconnect the esp8266 and
rs485 board for the measurements to only inspect the pool.

The resistance especially between the data lines should be around 120 Ohms.
In this case, please remove R7 from the module as it is a superfluous
additional termination resistor.

Good luck and thanks in advance for your feedback
Sascha