cribskip/esp8266_spa #36: M5 stack help

Doser86, 2024-09-26:
Anyone have a how to and yaml for setting it up on a atom m5 stack? 
Have went thru ewerything i feel with no luck. I want to have it only in ESPHome

DadMasterFlex, 2025-04-28:
I know your post is a few months old, but did you ever get this figured out?

Warrior700, 2025-04-30:
I also use an M5 Atom lite [https://docs.m5stack.com/en/core/ATOM%20Lite](url) with an RS485 shield [https://docs.m5stack.com/en/atom/Atomic%20RS485%20Base](url)

The RS485 shield's input voltage isn't limited to +12 V. The SPA mainboard +15.9 V is acceptable. The built-in A0Z1282CI DC/DC converter offers an input voltage range of 4.5 V to 36 V.

**BUT CAUTION: Check the correct pins for + and GND with a multimeter. Otherwise, you could destroy your Atom or possibly your SPA board. (In the worst case scenario.)** picture below

In my first attempts, I only connected A and B . The M5 was powered by a USB powerbank. After that worked, I connected GND and + to the M5/RS485 shield.

I used this cable to connect to the SPA motherboard:  [https://www.amazon.de/dp/B00S6AEMN2?ref_=ppx_hzsearch_conn_dt_b_fed_asin_title_1](url) for this: the red cable is GND, unfortunately, and the yellow one + V but check it with multimeter!!!! A and B are the two black

this is my yaml

```
esphome:
  name: "balboaspa"
  friendly_name: balboa_spa
  #min_version: 2024.10.0
  name_add_mac_suffix: false
  includes:
    - spa_reader.h

  libraries:
    - rlogiacco/CircularBuffer

 

esp32:
  board: m5stack-atom
  framework: 
    type: arduino
    version: latest
    

external_components:
  - source:
      type: git
      url: https://github.com/robertklep/esphome-custom-component
    components: [ custom, custom_component ]

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption: 
    key: "insert your key"

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  id: spa_uart_bus
  tx_pin: GPIO19
  rx_pin: GPIO22
  data_bits: 8
  parity: NONE
  stop_bits: 1
  baud_rate: 115200
  rx_buffer_size: 128

sensor:
- platform: custom
  lambda: |-
    auto spareader = new SpaReader(id(spa_uart_bus));
[... 727 lines cut ...]
        } else if (Q_in[2] == id && Q_in[4] == 0x23) { // FF AF 23:Filter Cycle Message - Packet index offset 5
          if (last_state_crc != Q_in[Q_in[1]]) {
            ESP_LOGD("Spa/debug/have_faultlog", "decoding filter settings");
            decodeFilterSettings();
          }
        } //else {
          // DEBUG for finding meaning
          //if (Q_in[2] & 0xFE || Q_in[2] == id)
          //print_msg(Q_in);
        //}

        // Clean up queue
        yield();
        Q_in.clear();
      }
      lastrx = millis();
    }
  }
};

``` 

i copied the spa_reader.h in the directory to HA  \config\esphome\spa_reader.h

[image]

Doser86, 2025-04-30:
> I know your post is a few months old, but did you ever get this figured out?

No. I have trouble finding rx and tx. I have 5 boards at home now that don't work with my spa wiring 😂