cribskip/esp8266_spa #37: Connected to wifi... but no data

heinekmadsen, 2025-03-28:
Hi...

So I have been trying to get this to work with my balboa control box in my spa for several hours/days now.

I'm using a Wemos D1 Mini Pro with a RS485 to TTL ( https://www.amazon.it/RS485-adattatore-Serial-Level-Converter-Module/dp/B0716VF1CC )

Wiring:
DC-DC converter set to output 3.3v
Wemos TX to RS485 TX
Wemos RX to RS485 RX
Wemos GND + RS485 GND + DC-DC output GND
RS485 A to SPA A
RS485 B to SPA B

I have commented out the hard reset lines in the code.
I can verify that it connects to Wifi AND it connects to my MQTT broker.

However... i no data i received from the SPA.

I have tried to find the solution in the issues "archive" but no luck.

any advise is much appreciated.

cribskip, 2025-03-28:
Hi, do you get readings on Spa/node/msg? There's a loop reading the
messages, maybe debug there and print/transfer the received data by mqtt.

Best,
Sascha

heinekmadsen, 2025-03-28:
Well... All i get i mqtt is this:

[image]

I found a fork of your repo where the readme states that it has to be a MAX13487 chip on the TTL converter.
I'm not sure if mine has a MAX485 or a MAX13487.

heinekmadsen, 2025-03-28:
Tried zooming with my phone on the chip... i can barely see it... but what i can see looks like its a MAX485